RavenLRS/raven #1: No OLED after flashing from stock

brycedjohnson, 2018-05-08:
I have two 915 modules that the OLED doesn't work after flashing from stock. The Arduino OLED_LoRa_Sender project works with with the OLED which use the same pins as raven (4, 15, 16 and i2c address of 0x3c).  On my other 2 modules of a different brand - the Oled does work.

[image]

[image]
[image]

Brand that OLED works:
[image]

```$ make monitor
MONITOR
--- idf_monitor on COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x000,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
I (28) boot: ESP-IDF v3.0-dirty 2nd stage bootloader
I (28) boot: compile time 17:45:00
I (29) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factoory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x36204 (221700) map
I (169) esp_image: segment 1: paddr=0x0004622c vaddr=0x3ffc0000 size=0x059e4 ( 23012) load
I (178) esp_image: segment 2: paddr=0x0004bc18 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at F:/msys32/home/windo/esp/esp-idf/components/freertos/xtensa_
vectors.S:1685

I (179) esp_image: segment 3: paddr=0x0004c020 vaddr=0x40080400 size=0x03ff0 ( 16368) load
I (194) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x99318 (627480) map
0x400d0018: _flash_cache_start at ??:?

I (416) esp_image: segment 5: paddr=0x000e9338 vaddr=0x400843f0 size=0x13574 ( 79220) load
0x400843f0: update_written_pages at F:/msys32/home/windo/esp/esp-idf/components/spi_flash/fl
ash_mmap.c:318
 (inlined by) spi_flash_ensure_unmodified_region at F:/msys32/home/windo/esp/esp-idf/compone
nts/spi_flash/flash_mmap.c:344

I (449) esp_image: segment 6: paddr=0x000fc8b4  vaddr=0x400c0000 size=0x00064 (   100) load
I (463) boot: Loaded app from partition at offset 0x10000
I (463) boot: Disabling RNG early entropy source...
I (464) cpu_start: Pro cpu up.
I (468) cpu_start: Starting app cpu, entry point is 0x400810d4
0x400810d4: call_start_cpu1 at F:/msys32/home/windo/esp/esp-idf/components/esp32/cpu_start.c
:215

I (0) cpu_start: App cpu up.
I (478) heap_init: Initializing. RAM available for dynamic allocation:
I (485) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (491) heap_init: At 3FFD6578 len 00009A88 (38 KiB): DRAM
I (497) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (503) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (510) heap_init: At 40097964 len 0000869C (33 KiB): IRAM
I (516) cpu_start: Pro cpu start user code
I (199)  cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (324) config: Module addr is 63:9D:0C:E9:A5:51
I (327) wifi: wifi firmware version: 6450435
I (327) wifi: config NVS flash: enabled
I (327) wifi: config nano formating: disabled
I (331) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (340) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (368) wifi: Init dynamic tx buffer num: 32
I ((368) wifi: Init data frame dynamic rx buffer num: 32
I (368) wifi: Init management frame dynamic rx buffer num: 32
I (372) wifi: wifi driver task: 3ffe045c, prio:23, stack:4096
I (377) wifi: Init static rx buffer num: 10
I (381) wifi: Init dynamic rx buffer num: 32
I (385) wifi: wifi power manager task: 0x3ffe540c prio: 21 stack: 2560
I (391) wifi: set country: cc=XXX schan=1 nchan=14 policy=1

W (3988) wifi: ap start fail

I (400) Settings: Setting bind changed
I (404) BTDM_INIT: BT controller compile version [8c18895]

I (410) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (475) LoRa: Got LoRa chip version 18
I (478) RC: Reconfigure input
I (482) RC: Starting bind
I (485) Output.Air.Bind: Start bind
I (676) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 1, 0
I (764) wifi: mode : null
I (765) wifi: ic_enable_sniffer
I (818) screen.i2c: u8g2_InitDisplay
I (818) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:
0
I (824) screen.i2c: u8g2_SetPowerSave
```

fiam, 2018-05-08:
@brycedjohnson is that log from the one that works or from the one that doesn't?

brycedjohnson, 2018-05-08:
@fiam: The one that doesn't work.

fiam, 2018-05-08:
@brycedjohnson Try with the code I've just pushed, please. I added a log message indicating whether Raven detects the OLED. If it says it doesn't detect it, edit `main/ui/screen_i2c.c` and change:

`#define SCREEN_I2C_MASTER_FREQ_HZ 1800000`

To

`#define SCREEN_I2C_MASTER_FREQ_HZ 1000000`

Let me know if that fixes the problem.

brycedjohnson, 2018-05-08:
I actually tried switching that to 
`#define SCREEN_I2C_MASTER_FREQ_HZ 100000` (one less 0 than yours) yesterday, but I just tried 
'#define SCREEN_I2C_MASTER_FREQ_HZ 1000000' and '#define SCREEN_I2C_MASTER_FREQ_HZ 1800000'
At all 3 of those speeds I got
`I (823) UI: Screen detected`

but nothing on the screen.

I know probably a pain because all these vendors are using slightly different parts and layouts.  I had noticed when I was looking into this ttgo v2 that had a different pinout from the others as well! (I don't have v2 though)

closed 2018-05-08 by fiam

Benzel, 2018-07-27:
I alsof have this issue on the RX oled is working but if i flash the TX the screen is not working. Is there already a solution?

commit 57402f34 (2018-05-08): Make the lowest contrast setting use 1 rather than 0

0 makes some boards turn the screen off completely. Thanks
to @brycedjohnson for noticing the problem and helping debug
the it until we found a fix.

Fixes #1