RavenLRS/raven #2: Can't Bind my two 868/915 modules

brycedjohnson, 2018-05-08:
Both modules should be their defaults.  I had done a make erase_flash on them.  Plugging in my 3rd module - (2nd tx) that the oled isn't working (#1) I can see another pop up on p2p in the corner and in the make monitor, but no bind option.  

Also I did verify that with the OLED_LoRa_Sender and OLED_LoRa_Receiver arduino programs that the are able to send and receive.

[image]

RX make monitor then plugging in TX
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip:: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
I (28) boot: ESP-IDF v3.0-dirty 2nd stage bootloader
I (28) boot: compile time 09:11:55
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x362ac (221868) map
I (171) esp_image: segment 1: paddr=0x000462d4 vaddr=0x3ffc0000 size=0x05b6c ( 23404) load
I (181) esp_image: segment 2: paddr=0x0004be48 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at F:/msys32/home/windo/esp/esp-idf/components/freertos/xtensa_
vectors.S:1685

I (182) esp_image: segment 3: paddr=0x0004c250 vaddr=0x40080400 size=0x03dc0 ( 15808) load
I (196) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x9a964 (633188) map
0x400d0018: _flash_cache_start at ??:?

I (420) esp_image: segment 5: paddr=0x000ea984 vaddr=0x400841c0 size=0x139a8 ( 80296) load
0x400841c0: spi_flash_disable_interrupts_caches_and_other_cpu at F:/msys32/home/windo/esp/es
p-idf/components/spi_flash/cache_utils.c:115

I (454) esp_image: segment 6: paddr=0x000fe334 vaddr=0x400c0000 size=0x00064 (   100) load
I (468) boot: Loaded app from partition at offset 0x10000
I (468) boot: Disabling RNG early entropy source...
I (469) cpu_start: Pro cpu up.
I (472) cpu_start: Starting app cpu, entry point is 0x400810e8
0x400810e8: call_start_cpu1 at F:/msys32/home/windo/esp/esp-idf/components/esp32/cpu_start.c
:215

I (0) cpu_start: App cpu up.
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (496) heap_init: At 3FFD6700 len 00009900 (38 KiB): DRAM
I (502) heapp_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (515) heap_init: At 40097B68 len 00008498 (33 KiB): IRAM
I (521) cpu_start: Pro cpu start user code
I (203) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (302) config: Module addr is 88:2E:5F:D5:4B:44
I (304) wifi: wifi firmware version: 6450435
I (304) wifi: config NVSS flash: enabled
I (304) wifi: config nano formating: disabled
I (308) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (318) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (356) wifi: Init dynamic tx buffer num: 32
I (356) wifi: Init data frame dynamic rx buffer num: 32
I (356) wifi: Init management frame dynamic rx buffer num: 32
I (360) wifi: wifi driver task: 3ffd8a6c, prio:23, stack:4096
I (365) wifi: Init static rx buffer num: 10
I (369) wifi: Init dynamic rx buffer num: 32
I (373) wifi: wifi power manager task: 0x3ffd8bd4 prio: 21 stack: 2560
I (379) wifi: set country: cc=XXX schan=1 nchan=14 policy=1

W (386) wifi: ap start fail

I (388) Settings: Setting bind changed
I (392) BTDM_INIT: BT controller compile version [8c188895]

I (399) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (463) LoRa: Got LoRa chip version 18
I (466) RC: Reconfigure input
I (471) RC: Starting bind
I (471) RC: Input invalidated
I (472) RC: Reconfigure input
I (473) Input.Air.Bind: Open
I (699) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 1, 0
I (793) wifi: mode : null
I (794) wifi: ic_enable_sniffer
I (847) screen.i2c: u8g2_InitDisplay
I (847) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:
0
I (854) screen.i2c: u8g2_SetPowerSave
I (9690) RMP: Added p2p peer (can authenticate: N)

```

TX make monitor - then plugging in RX
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv::0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:0
load:0x400788000,len:13756
entry 0x40078fb4
I (28) boot: ESP-IDF v3.0-dirty 22nd stage bootloader
I (28) boot: compile time 17:45:00
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 sizze=0x36204 (221700) map
I (171) esp_image: segment 1: paddr=0x0004622c vaddr=0x3ffc0000 size=0x059e4 ( 23012) load
I (181) esp_image: segment 2: paddr=0x0004bc18 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at F:/msys32/home/windo/esp/esp-idf/components/freertos/xtensa_
vectors.S:1685

I (181) esp_image: segment 3: paddr=0x0004c020 vaddr=0x40080400 size=0x03ff0 ( 16368) load
I (196) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x99318 (627480) map
0x400d0018: _flash_cache_start at ??:?

I (418) esp_image: segment 5: paddr=0x000e9338 vaddr=0x4000843f0 size=0x13574 ( 79220) load
0x400843f0: spi_flash_ensure_unmodified_region at F:/msys32/home/windo/esp/esp-idf/component
s/spi_flash/flash_mmap.c:318

I (451) esp_image: segment 6: paddr=0x000fc8b4 vaddr=0x400c0000 size=0x00064 (   100) load
I (465) boot: Loaded app from partition at offset 0x10000
I (466) boot: Disabling RNG early entropy source...
I (466) cpu_start: Pro cpu up.
I (470) cpu_start: Starting app cpu, entry point is 0x400810d4
0x400810d4: start_cpu1_default at F:/msys32/home/windo/esp/esp-idf/components/esp32/cpu_star
t.c:387

I (0) cpu_start: App cpu up.
I (480) heap_init: Initializing. RAM available for dynamic allocation:
I (487) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (493) heap__init: At 3FFD6578 len 00009A88 (38 KiB): DRAM
I (499) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (505) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (512) heap_init: At 40097964 len 0000869C (33 KiB): IRAM
I (518) cpu_start: Pro cpu start user code
I (201) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (302) config: Module addr is 8B:E1:4B:7A:0F:6D
I (305) wifi: wifi firmware version: 6450435
I (305) wifi: config NVS flash: enabled
I (305) wifi: config nano formating: disabled
I (309) system_api: Base MAC address is not set, read default bbase MAC address from BLK0 of
 EFUSE
I (318) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (355) wifi: Init dynamic tx buffer num: 32
I (356) wifi: Init data frame dynamic rx buffer num: 32
I (356) wifi: Init management frame dynamic rx buffer num: 32
I (359) wifi: wifi driver task: 3ffdf02c, prio:23, stack:4096
I (365) wifi: Init static rx buffer num: 10
I (368) wifi: Init dynamic rx buffer num: 32
I  (372) wifi: wifi power manager task: 0x3ffe3a64 prio: 21 stack: 2560
I (379) wifi: set country: cc=XXX schan=1 nchan=14 policy=1

W (385) wifi: ap start fail

I (388) Settings: Setting bind changed
I (391) BTDM_INIT: BT controller compile version [8c18895]

I (398) system_api: Base MAC address is not set, read default base MAC address from BLK0 of
EFUSE
I (462) LoRa: Got LoRa chip version 18
I (465) RC: Reconfigure input
I (469) RC: Starting bind
I (472) Output.Air.Bind: Start bind
I (672) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 1, 0
I (761) wifi: mode : null
I (761) wifi: ic_enable_sniffer
I (815) screen.i2c: u8g2_InitDisplay
I ((815) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr
:0
I (822) screen.i2c: u8g2_SetPowerSave
I (11276) RMP: Added p2p peer (can authenticate: N)

fiam, 2018-05-10:
After doing some tests, trying to bind with one of these modules as TX and another differently branded one (verified to work properly) as RX makes the RX see the bind, but it doesn't complete because the confirmation packet from the RX doesn't make it to the TX.

fiam, 2018-10-09:
I guess at this point we can assume it was faulty hardware, since newer TTGO boards are working fine on 868 and 915.

closed 2018-10-09 by fiam