RavenLRS/raven #7: Design a PCB

SwiftFlying, 2018-06-21:
Can you tell me your mailbox address?I want to discuss some problems with you and redesign a PCB.

hdphilip, 2018-06-24:
what kind of issues are you having?

fiam, 2018-06-24:
@SwiftFlying Thanks for your interest! We're doing most of the discussion in INAV's slack in the #lora-baby channel, but the auto-invite system is currently down, so we'd need your email to send you an invite.

SwiftFlying, 2018-06-25:
I want to design a very small PCB to run the RavenLRS firmware. My mailbox address is 759315223@qq.com

jonathan2018, 2018-06-29:
RavenLRS  as whats i saw it is opensource . This guy want to Sell  , please do not do that

pheex, 2018-07-06:
Hi, @fiam can you send me an invite, i'm working on an esp8266+lora module myself, but i'll eventually switch to esp32 in near future, ravenlrs seemed very promising.
3qau0why7djyrxr@jetable.org

fiam, 2018-07-06:
@pheex @SwiftFlying  I've set up a telegram channel for Raven: https://t.me/RavenLRS