RavenLRS/raven #19: Docker image is supported

doskoi, 2018-12-13:
I have created a docker image that can easily to setup build environment, the repo is here: https://github.com/doskoi/RavenLRS

It has been tested on Docker CE on macOS Mojave and Windows 10 Pro, Linux Mint 19 (apt install docker.io)

I'm glad to have more tests, so I send it here for experienced users to send any feedback.

doskoi, 2019-01-08:
@fiam You can use docker image as same as [inav](https://github.com/iNavFlight/inav/) to put a build.sh in project, the script is accpet build target as parameter to passthrough to the make CLI.

fiam, 2019-01-08:
@doskoi This is definitely very nice. Would you like to submit a PR or would you prefer that I link to your repo from the README?

doskoi, 2019-01-09:
Sure, I will send a PR with build script for build firmware with docker,
and feel free to update README for advanced usage. Like flash, monitor, use
git to update repo. The examples already in my README.

doskoi, 2019-01-13:
I've made the update and tested, now it works via build.sh in raven folder or initialize project via docker commands in empty folder.

Pull request: https://github.com/RavenLRS/raven/pull/40