RavenLRS/raven #21: LoRa reliability for RC control

vkolotov, 2019-01-02:
I find this project is very interesting. A very good idea, well done! I'd like to try this out. However, before I started testing/building it, I'd like to clarify something.

I've heard that LoRa is designed for long range communication mainly for IoT devices, where reliability  or, another words, frame drops are not important. Hence, a logical question arises, is it really can be used for controlling drones?

Have you guys done any reliability tests?

Cheers.

DieHertz, 2019-01-02:
Not meaning to answer your question, are you aware that TBS Crossfire and FrSky R9 are based on the same technology?

vkolotov, 2019-01-02:
I was not aware that it is based on LoRa. I know that it is based on 8xx/9xx mhz ranges and uses crossfire protocol. Does raven uses the same crossfire protocol?

DieHertz, 2019-01-02:
All three are based on SemTech chips,  SX1276 or similar. These chips are informally called LoRa. LoRa is also the name for the long-range low-bandwidth modulation scheme available on them, which gives the maximum range (and minimum update-rate).

CRSF is just a serial protocol and isn't really defining anything as far as RF design is concerned.
RavenLRS Tx and Rx can input/output CRSF among other protocols.

vkolotov, 2019-01-02:
Great! Thank you. Any recommendations on a dev/testing board (preferably from aliexpress)? As a drone racer weight/size is important for me, I was looking something small/light but found only 1 module that supports only 433mhz range which I cannot use in New Zealand...

vkolotov, 2019-01-02:
As I said above, I'm more interested in low latency rather than range. Is this something that LoRa / Raven can provide? I see it supports 100hz, which is very promising.

I'm planning to use Taranis Q X7. I know that there are some issues with internal module latency, does it affect Raven as well? Just trying to understand if declared 100hz is what I'm going to get with X7 and Raven. Cheers

fiam, 2019-01-10:
@vkolotov Both Crossfire and the FrSky R9 line use LoRa for communication, so I'd say the hardware is more than proven to work.

The low bitrate on the SmartPort line in the Q X7 (with the additional delay it introduces) affects Raven, but it happens in everything that uses that line to communicate with the radio. At some point I might design a custom protocol using 2 lines to communicate with the radio (which will eliminate those problems on Q X7, Horus, etc...), but right now there are other things with higher priority.

closed 2019-01-10 by fiam