RavenLRS/raven #45: Enhance failsafe options for Channels (PWM) output

nmaggioni, 2019-02-21:
The current F/S behavior for the direct PWM output protocol is equivalent to OpenTX's "Hold": the outputs aren't updated until the TX reconnects. This may be dangerous for crafts that use Raven without an FC.

It would be useful if F/S options mirrored the ones [canonically available on radios][OTX_FS]: **Hold**, **No Pulses** (or, better yet, _THR to 0% and all the other channels centered_), **Custom** (_current sticks position when setting F/S_).

The RX settings could be expanded to include additional entries when the **Channels** or **SBUS** (for the sake of completeness) protocols are selected:

+ F/S Mode:
  - Hold
  - No Pulses
    + THR channel: CH1-16
  - Custom
    + Set current values >>

[OTX_FS]: https://opentx.gitbooks.io/manual-for-opentx-2-2/model_setup.html

kilrah, 2019-02-21:
Why make "no pulses" something fixed that can already be set up with Custom instead of being a real "no pulses"?

nmaggioni, 2019-02-21:
@kilrah You are right, it's not needed. I was mapping modes 1:1 but they can be simplified.

kilrah, 2019-02-21:
What I meant is that No Pulses is a ueful choice (although in increasingly rare cases) so if you were going to add it then it should be just that and not a "hardcoded custom".

closed 2019-06-13 by nmaggioni